ghi-electronics/Documentation #176: What string formatting is supported?

gus-ghielec, 2020-05-11:
We need to document what formatting is supported. For example, is this a bug or not supported?

https://forums.ghielectronics.com/t/are-spi2-and-spi5-available-on-sc20260d-dev-board/22974/8
'''
private static void BugForGus()
        {
            for (Double i = 0.0; i < 10.0; i += 0.1)
            {
                Debug.WriteLine($"i = {i:F1}");
                Thread.Sleep(50);
            }
        }
'''

JoelGHI, 2020-05-13:
This is a bug. The following code (same as above except it prints three decimal places instead of one) runs to completion without an exception; however there are rounding errors (0.801 instead of 0.800, 1.201 instead of 1.200).

`private static void BugForGus() {
    for (Double i = 0.0; i < 10.0; i += 0.1) {
        System.Diagnostics.Debug.WriteLine($"i = {i:F3}");
        Thread.Sleep(50);
    }
}`

gus-ghielec, 2020-05-13:
Still, we need to document what is supported. Even if it is briefly.

PR #198 (2020-05-26): Added info on supported ToString() arguments.
Fixes #176
closed 2020-05-26 by gus-ghielec