ghi-electronics/Documentation #197: Max SPI frequencies

MikroBusNet, 2020-05-26:
Hello,

Could you add somewhere the maximum allowable frequencies for the different SPI buses, please ?
I presume that they are not all on the same prescalers.

MikroBusNet, 2020-05-27:
FYI, the values returned by the method do not seem to be correct.

On the SC20260D dev board, I get the following :

> SPI2 min freq 468750, max freq 60000000
> SPI3 min freq 468750, max freq 60000000
> SPI5 min freq 468750, max freq 60000000

However, in the following [post on the forum](https://forums.ghielectronics.com/t/are-spi2-and-spi5-available-on-sc20260d-dev-board/22974/6?u=bec_a_fuel) I've seen that :

> Requested 30 MHz is ok on SPI3
> Requested 24 MHz is ok on SPI2
> Requested 14 MHz is ok on SPI5
> 

Higher frequencies on SPI2 and SPI5 did not work.

I was using a device that can handle 30MHz max.

Palomino34, 2020-05-27:
it has been corrected on Preview7.

SPI123 max 30MHz
SPI45 max 24MHz.

MikroBusNet, 2020-05-27:
Is this a "board" limitation (SC20100 vs SC20260, for example) or a chip limitation ?
Because somewhere else in the docs you say that the max frequency goes up to 50 MHz.

PR #199 (2020-05-28): Added info about max SPI clock speed.
Fixes #197
closed 2020-05-28 by gus-ghielec