ghi-electronics/Documentation #351: Provide default USB PID

gus-ghielec, 2020-11-09:
GHI will provide a default (and licensed for commercial use) USB PIDs for CDC and WinUSB usage on GHI's SITCore products. The library should just include them internally but the user can override them if they need. The customer should NOT modify the PID unless they are modifying the VID to what they own or they are licensed to use. This must be very clear in the docs.

GHI VID 0x1B9F
GHI licensed CDC PID 0xF001
GHI licensed WinUSB PID 0xF002

Palomino34, 2020-11-25:
GHI VID 0x1B9F

USBClient PID:
Mouse = 0xF000,
CDC = 0xF001,
MassStorage = 0xF002,
Keyboard = 0xF004,
Joystick = 0xF005,
WinUsb = 0xF006,

Bootloader 2.x PID:
0x0104

TinyCLR 2.x PID:
0x5010

closed 2020-11-25 by gus-ghielec

Palomino34, 2020-12-04:
Update:

GHI VID 0x1B9F

USBClient PID:
Mouse = 0xF000,
CDC = 0xF001,
MassStorage = 0xF002,
Keyboard = 0xF004,
Joystick = 0xF005,
WinUsb = 0xF006,
RawDevice= 0xF007,

Bootloader 2.x PID:
0x0104

TinyCLR 2.x PID:
0x5010