ghi-electronics/Documentation #404: Fix network IP address even check

Palomino34, 2021-01-08:
```
if (address[0] != 0 && address[1] != 0 && address[2] != 0 && address[3] != 0)
                LinkReady = true;

```

should be changed to 

```
if (address[0] != 0 || address[1] != 0 || address[2] != 0 || address[3] != 0)
                LinkReady = true;

```

poundy, 2021-01-08:
how will this work if the most universal address space is used on a home network, 192.168.0.x, when the device gets an address of 192.168.0.14 ?

gus-ghielec, 2021-01-08:
Zero is allowed in the IP address but they shouldn't all be zero

poundy, 2021-01-10:
oh yeah. I should have checked the logic better, it's clear if any is non-zero then the link address has changed - in fact the problem i describe is the reason the original code didn't work

greg-norris, 2021-01-19:
Should this be fixed on DOCS or LIBS?

gus-ghielec, 2021-02-04:
didn't find this anywhere on docs!

closed 2021-02-04 by gus-ghielec