ghi-electronics/Documentation #461: Azure: Generate SAS

Palomino34, 2021-02-19:
use this file below to generate SAS when document:

SharedAccessSignatureBuilder.cs:
```
//------------------------------------------------------------------------------ 
// Copyright (C) 2021 GHI Electronics
//
// This file is a modified version from Microsoft.
//
//------------------------------------------------------------------------------

/// <summary>
/// Builds Shared Access Signature (SAS) tokens.
/// </summary>
public class SharedAccessSignatureBuilder
{
    private string _key;

    /// <summary>
    /// Initializes a new instance of <see cref="SharedAccessSignatureBuilder"/> class.
    /// </summary>
    public SharedAccessSignatureBuilder()
    {
        TimeToLive = TimeSpan.FromMinutes(60);
    }

    /// <summary>
    /// The shared access policy name.
    /// </summary>
    public string KeyName { get; set; }

    /// <summary>
    /// The shared access key value.
    /// </summary>
    public string Key
    {
        get => _key;

        set
        {
            // TQD StringValidationHelper.EnsureBase64String(value, "Key");
            _key = value;
        }
    }

    /// <summary>
    /// The resource Id being accessed.
    /// </summary>
    public string Target { get; set; }

    /// <summary>
    /// The time the token expires.
    /// </summary>
    public TimeSpan TimeToLive { get; set; }

    /// <summary>
    /// Build a SAS token.
    /// </summary>
    /// <returns>SAS token.</returns>
    public string ToSignature()
    {
        return BuildSignature(KeyName, Key, Target, TimeToLive);
    }

    private string BuildSignature(string keyName, string key, string target, TimeSpan timeToLive)
    {
        string expiresOn = BuildExpiresOn(timeToLive);
        string audience = WebUtility.UrlEncode(target);            
        
        // Example string to be signed:
        // dh://myiothub.azure-devices.net/a/b/c?myvalue1=a
        // <Value for ExpiresOn>
        var request = audience + "\n" + expiresOn;

        string signature = Sign(request, key);

        // Example returned string:
        // SharedAccessSignature sr=ENCODED(dh://myiothub.azure-devices.net/a/b/c?myvalue1=a)&sig=<Signature>&se=<ExpiresOnValue>[&skn=<KeyName>]

        var buffer = new StringBuilder();            

        buffer.Append(string.Format("{0} {1}={2}&{3}={4}&{5}={6}",
         MySharedAccessSignatureConstants.SharedAccessSignature,
            MySharedAccessSignatureConstants.AudienceFieldName, audience,
          MySharedAccessSignatureConstants.SignatureFieldName, WebUtility.UrlEncode(signature),
        MySharedAccessSignatureConstants.ExpiryFieldName, WebUtility.UrlEncode(expiresOn)));            

        if (!IsNullOrWhiteSpace(keyName))
        {                
            buffer.Append(string.Format("&{0}={1}",
                MySharedAccessSignatureConstants.KeyNameFieldName, WebUtility.UrlEncode(keyName)));
        }

        return buffer.ToString();
    }

    private static string BuildExpiresOn(TimeSpan timeToLive)
    {
        DateTime expiresOn = DateTime.UtcNow.Add(timeToLive);            
        TimeSpan secondsFromBaseTime = expiresOn.Subtract(MySharedAccessSignatureConstants.EpochTime);
        long seconds = (Int64)secondsFromBaseTime.TotalSeconds; 
        return seconds.ToString();
    }

    /// <summary>
    /// Sign the request string with a key.
    /// </summary>
    /// <param name="requestString">The request string input to sign.</param>
    /// <param name="key">The secret key used for encryption.</param>
    /// <returns>The signed request string.</returns>
    protected virtual string Sign(string requestString, string key)
    {            
        var algorithm = new HMACSHA256(Convert.FromBase64String(key));

        return Convert.ToBase64String(algorithm.ComputeHash(Encoding.UTF8.GetBytes(requestString)));
    }

    private bool IsNullOrWhiteSpace(string s)
    {
        if (s == null)
            return true;

        if (s.IndexOf(" ") >= 0)
            return true;

        return false;
    }
}

static class MySharedAccessSignatureConstants
{
    public const int MaxKeyNameLength = 256;
    public const int MaxKeyLength = 256;
    public const string SharedAccessSignature = "SharedAccessSignature";
    public const string AudienceFieldName = "sr";
    public const string SignatureFieldName = "sig";
    public const string KeyNameFieldName = "skn";
    public const string ExpiryFieldName = "se";
    public const string SignedResourceFullFieldName = SharedAccessSignature + " " + AudienceFieldName;
    public const string KeyValueSeparator = "=";
    public const string PairSeparator = "&";

    public static readonly DateTime EpochTime = new DateTime(1970, 1, 1, 0, 0, 0, 0);
    public static readonly TimeSpan MaxClockSkew = new TimeSpan(0, 5, 0);
}
```

greg-norris, 2021-02-24:
Documentation updated, need to test SAS Nuget

Palomino34, 2021-02-24:
I tested!

closed 2021-02-24 by Palomino34