ghi-electronics/Documentation #495: add resistive touch

Palomino34, 2021-03-05:


Palomino34, 2021-03-05:
```cs
           var touch = new ResistiveTouchController(
                320, // Screen width
                240, // Screen height
                SC20260.GpioPin.PA0,  // digital pin support analog
                SC20260.GpioPin.PA3,  // digital pin support analog
                SC20260.GpioPin.PA5,  // digital pin
                SC20260.GpioPin.PC3,  // digital pin
                SC20260.Adc.Controller1.Id, // Analog controller id
                SC20260.Adc.Controller1.PA0, // analog channel id
                SC20260.Adc.Controller1.PA3 // analog channel id
                );

            touch.ScaleX = new Scale(20, 280); // option to get more accurate point
            touch.ScaleY = new Scale(20, 200); // option to get more accurate point

            while (true)
            {
                Thread.Sleep(100); // poll interval every 100ms

                var x = touch.X;
                var y = touch.Y;

                if (x >= 0 && y >= 0) // detected touch
                    Debug.WriteLine("X: " + x +", Y = " + y);
            }
```

greg-norris, 2021-03-05:
Added in PR #496

closed 2021-03-05 by greg-norris