ghi-electronics/Documentation #505: TLS Tutorial should be extended with Server part

AndreMarschalek, 2021-03-15:
the docs should be splitted into Client and Server part
where the Server part should contain:

1. a way to create self signed certificates
2. how to import the root certificate for trust (with screenshots or not..)
3. how to show or remove certificates from PC (with screenshots or not..)
3. sample code how to use X509Certificate
4. sample code how to create SslStream object

AndreMarschalek, 2021-03-15:
1: many ways exists to reach the same goal, my way is based on Linux and [OpenSSL](https://www.openssl.org/) and creates not only a Server Certificate, but also a CA Certificate
an existing CA Certificate makes it easier the Client PC trusts the Certificates as he only need to trust the CA Certificate and all Certificates signed by this CA (your own CA) are automatically trusted
this not only makes it easier for testing purposes if you need to create more Certificates

to make this process easier [files.zip](https://github.com/ghi-electronics/Documentation/files/6140988/files.zip) includes 3 files
- ca.cnf 
ca certificate configuration
- server.cnf
server certificate configuration
- server_v3.ext
server certificate extension

Steps (do all this in the zip extraction folder)
- create CA (Certificate Authority)
1. edit ca.cnf
change the values for C, ST, L and O
2. Generate a CA private key
```$ openssl genrsa -out ca_privatekey.pem 2048```
3. Generate a 10 years valid CA certificate (from Linux Shell)
```$ openssl req -new -x509 -days 3650 -nodes -key ca_privatekey.pem -sha256 -out ca.crt -config ca.cnf```

this produced a file named **ca.crt** you can publish and import in your PC

- create Server Certificate
1. edit server.cnf
change the values for C, ST, L and O
change the value of CN, with DNS the FQDN (server.exampledomain.local), NetBIOS Name (SERVER) or IP Address
2. edit server_v3.ext
change the values in alt_names section, important to have an entry for the CN value from before
important to include CN content from step before!
3. Create a Certificate Signing Request
```$ openssl req -new -nodes -out server.csr -keyout serverkey.key -config server.cnf```
4. Generate a 1 year valid self signed Server Certificate
```$ openssl x509 -req -in server.csr -CA ca.crt -CAkey ca_privatekey.pem -CAcreateserial -out servercert.crt -days 365 -extfile server_v3.ext```

this produced 2 files named **serverkey.key** and **servercert.crt** you need to use in your device

AndreMarschalek, 2021-03-15:
2: install the CA Certificate in "Current User" Store Location and "Trusted Root Certification Authorities" Certificate store [screenshot_2.zip](https://github.com/ghi-electronics/Documentation/files/6141194/screenshot_2.zip)

1. double click on **ca.crt** and click "Install Certificate..."
2. be sure "Current User" is selected
3. select "Place all certificates in the following store" 
4. click on "Browse..." and select "Trusted Root Certitification Authorities"

from this moment your PC will trust all Certificates signed from your CA

AndreMarschalek, 2021-03-15:
3: [screenshot_3.zip](https://github.com/ghi-electronics/Documentation/files/6141285/screenshot_3.zip)

1. either run **certmgr.msc** or search for **cert** in Windows 10 Settings and select "Manage user certificates" to open Cert Manager for **Current User**, not for Local Computer
2. on left side select "Trusted Root Certification Authorities" and "Certificates"
3. on right side find your Certificate based on your values from **ca.cnf**

greg-norris, 2021-03-16:
Add in PR #513

closed 2021-03-16 by greg-norris