ghi-electronics/Documentation #518: Document about Basic graphic driver

Palomino34, 2021-03-19:
```cs
///
var basicGfx = new BasicGraphicDemo(1, 1);
var color1 = 0xFFF0000U;
var color2 = 0xFFFF000U;
var color3 = 0xFFFFF00U;

var cnt = 0;

while (true)
{

    basicGfx.Clear();
    basicGfx.DrawText("basic " + cnt++, color1, 5, 5 , 2, 1);

    basicGfx.DrawCircle(color2, 10, 10, 6);

    basicGfx.DrawRectangle(color3, 40, 40, 20, 30);

    basicGfx.Flush();

    Thread.Sleep(1);
}
///
```

BasicGraphicDemo.cs

```cs
using GHIElectronics.TinyCLR.Drivers.BasicGraphic;
using System;
using GHIElectronics.TinyCLR.Drivers.Sitronix.ST7735;
using GHIElectronics.TinyCLR.Devices.Spi;
using GHIElectronics.TinyCLR.Pins;
using GHIElectronics.TinyCLR.Devices.Gpio;

namespace BasicGraphicDemoTest 
{
    public class BasicGraphicDemo : BasicGraphic
    {
        private readonly byte[] buffer;

        private readonly ST7735Controller st7735;

        public int Width { get; }
        public int Height { get; }

        public int WidthMultiplier { get; }
        public int HeightMultiplier { get; }

        public BasicGraphicDemo() : this(1, 1) { }
        public BasicGraphicDemo(int widthMultiplier, int heightMultiplier)
        {
            if (widthMultiplier == 0 || heightMultiplier == 0)
                throw new ArgumentException("widthMultiplier, heightMultiplier can not smaller than 1.");

            // This demo runs on SC20100 Dev RevC
            var spi = SpiController.FromName(SC20100.SpiBus.Spi4);
            var gpio = GpioController.GetDefault();

            SpiConnectionSettings customConfig = ST7735Controller.GetConnectionSettings(SpiChipSelectType.Gpio, GpioController.GetDefault().OpenPin(SC20100.GpioPin.PD10));

            customConfig.ClockFrequency = 12_000_000;

            st7735 = new ST7735Controller(
                spi.GetDevice(customConfig), // ChipSelect 
                gpio.OpenPin(SC20100.GpioPin.PC4), // Pin RS
                gpio.OpenPin(SC20100.GpioPin.PE15)// Pin RESET                

            );

            var bl = gpio.OpenPin(SC20100.GpioPin.PA15); // back light

            bl.Write(GpioPinValue.High);
            bl.SetDriveMode(GpioPinDriveMode.Output);

            st7735.SetDataAccessControl(true, true, false, true); // rotate the screen
            st7735.SetDrawWindow(0, 0, 160, 128);

            this.WidthMultiplier = widthMultiplier;
            this.HeightMultiplier = heightMultiplier;

            this.Width = st7735.MaxWidth / this.WidthMultiplier;

            this.Height = st7735.MaxHeight / this.HeightMultiplier;
            this.buffer = new byte[this.Width * this.Height * 2];

            st7735.Enable();
        }

        public void Clear() => Array.Clear(this.buffer, 0, buffer.Length);
        public void Flush()
        {
            st7735.DrawBuffer(buffer, 0, 0, this.Width, this.Height, this.Width, this.WidthMultiplier, this.HeightMultiplier);
        }
        public override void SetPixel(int x, int y, uint color)
        {
            if (x < 0 || y < 0 || x >= this.Width || y >= this.Height) return;

            var idx = (y * this.Width + x) * 2;
            var clr = color;

            this.buffer[idx] = (byte)(((clr & 0b0000_0000_1111_1000_0000_0000_0000_0000) >> 16) | ((clr & 0b0000_0000_0000_0000_1110_0000_0000_0000) >> 13));
            this.buffer[idx + 1] = (byte)(((clr & 0b0000_0000_0000_0000_0001_1100_0000_0000) >> 5) | ((clr & 0b0000_0000_0000_0000_0000_0000_1111_1000) >> 3));
        }
    }
}
```

gus-ghielec, 2021-03-24:
Point kit y that this driver supports pixel multiplier. Meaning, a 160x128 display for example can be treated as 80x64 display, with 2,2 multiplier. This is a lower resolution but then it is 10k vs 40kb of needed ram.

greg-norris, 2021-04-07:
Added to Drivers Documentation in PR #517

closed 2021-04-07 by greg-norris