ghi-electronics/Documentation #530: IFU

greg-norris, 2021-04-08:


gus-ghielec, 2021-04-16:
And activity pin

Palomino34, 2021-04-20:
Use external flash to cache data:
```
var updater = new InFieldUpdate(QuadSpiController)
```

Use memory to cache data:
```
var updater = new InFieldUpdate();
```
Load application key:
```
updater.LoadApplicationKey(appKey);
```

Load application chunk:
```
updater.LoadApplicationChunk(data, 0, count);
```

Load firmware chunk:
```
updater.LoadFirmwareChunk(data, 0, count)
```

Verify:
```
var firmwareVersion = updater.VerifyFirmware();
var applicationVersion = updater.VerifyApplication();

```

Update Application from FileSystem directly:
```
 updater = new ApplicationdUpdate(filestream, appKey)
```
Verify:

```
var applicationVersion = updater.Verify();
```
Flash and Reset:

```
updater.FlashAndReset();
```

Palomino34, 2021-04-20:
Activity Pin:
```
var indicatorPin = GpioController.GetDefault().OpenPin(SC20260.GpioPin.PB0);

updater.ActivityPin = indicatorPin;

```

greg-norris, 2021-04-20:
Added in PR #557

closed 2021-04-20 by greg-norris